SysCV/LiDAR_snow_sim #4: Why did not `max_intensity` come from calibration file?

Beastmaster, 2022-06-27:
Hi, 
in file: `tools/snowfall/simulation.py`, line123 - 126:  
```
        if channel in [53, 55, 56, 58]:
            max_intensity = 230
        else:
            max_intensity = 255
```
Why did not `max_intensity` come from calibration file?

Thank you!

Beastmaster, 2022-06-27:
I'm also confused on "Algorithm1 LiDAR simulation", line25. 
[image]
Why add corrected  i_max to max received power(intnesity)?

MartinHahner, 2022-06-27:
> Why did not `max_intensity` come from calibration file?

It is the hard-coded version of what is in our calibration file. 😉

closed 2022-06-27 by MartinHahner

Beastmaster, 2022-06-28:
Do you have any clue to setup this `max_intensity`? :grin: