SysCV/LiDAR_snow_sim #5: Why is the beam divergence correction like that?

Beastmaster, 2022-06-27:
I can find nowhere about this equation (Chapter3, eq(2)): 

[image]

Correction factors are listed for each beam. Did I lose something?

MartinHahner, 2022-06-27:
From [the manual](https://www.manualslib.com/manual/1988532/Velodyne-Hdl-64e-S3.html):

[image]

closed 2022-06-27 by MartinHahner

yanconglin, 2024-08-02:
Hi, a followup question
why[ this line ](https://github.com/SysCV/LiDAR_snow_sim/blob/760c42e46e10c6aa6b57048c5c1892289f0c2d08/tools/snowfall/simulation.py#L128)does not align with the equation shown above?  I am referring to this: (1 - d_orig/120)**2.