SysCV/LiDAR_snow_sim #7: How to augment the KITTI bin file?

cuge1995, 2022-07-06:
Thanks for the great work!
The kitti bin only have x,y,z,intensity(not (x, y, z, intensity, channel)), how to augment it with snowfall in your paper?

MartinHahner, 2022-07-07:
Good question. 

You can either try to reconstruct the channel information geometrically using the vertical and horizontal resolution of the LiDAR sensor or you randomly use one of the 64 sampled snowflake disks for every point.

The first option is more sophisticated to implement.

The second option is easier to implement but will break consistency. 
Nevertheless, I think the snowfall clutter should not look drastically different that way.

closed 2022-08-21 by github-actions[bot]

dinghuixiang-github, 2022-11-28:
> 

 I have the same problem.Did you solve it? Could you please tell me how to achieve?