SysCV/LiDAR_snow_sim #8: How to add the snow+wet after running pointcloud_viewer.py? Q&A

cuge1995, 2022-07-07:
Only a `add our fog`, no `add our snow+wet`

MartinHahner, 2022-07-07:
Press the button `apply snow` and then the button `apply wet`. 😄

closed 2022-07-07 by MartinHahner

cuge1995, 2022-07-08:
I don't see any `apply snow` and `apply wet` button after running pointcloud_viewer.py.

cuge1995, 2022-07-08:
[image]
This is my window, what's wrong with me?

MartinHahner, 2022-07-08:
Which dataset did you load? KITTI?

If you want to augment KITTI, you have to make changes in the code.
The current code only works with the STF dataset.
It won't work out of the box for other datasets.

esther-ET, 2024-03-02:
Hello, Mr. Hahner, thank you so much for this project. I have the same problem just like the cuge1995's, but I have used the STF.  I use the STF by loading it through the button "choose custom directiry", and I have got the unzipped file called SeeingThroughFogCompressedExtracted with 12977 bin documents.

esther-ET, 2024-03-02:
Hi, @MartinHahner!
I have some question about the Gui, too. I can't apply the snow button and the wet button at the same time, and is it wrong? Also only when I choose the snow, I can apply snow+wet, but it doesn't look effective that the point cloud has no difference from the snow one after I load the result.