SysCV/LiDAR_snow_sim #12: What's the use of precompute.py

Rolandxx7, 2022-08-10:
Thanks for your great work! I wanna know what is the purpose of this file called precompute.py, because I do not see any corresponding note.

MartinHahner, 2022-08-10:
With [precompute.py](https://github.com/SysCV/LiDAR_snow_sim/blob/main/tools/snowfall/precompute.py), you can compute the _snowflake patterns_, which are then stored on disk. 
This process in its current implementation takes too much time to do on the fly while training your model. 
During training, you can then load those _snowflake patterns_ from the disk without computing them each time.

Rolandxx7, 2022-08-12:
> With [precompute.py](https://github.com/SysCV/LiDAR_snow_sim/blob/main/tools/snowfall/precompute.py), you can compute the _snowflake patterns_, which are then stored on disk. This process in its [...]

Thanks for your kind reply! Also I would like to know why do you use calibration of kitti dataset instead of dense dataset in precompute.py. Because you have said you 

> With [precompute.py](https://github.com/SysCV/LiDAR_snow_sim/blob/main/tools/snowfall/precompute.py), you can compute the _snowflake patterns_, which are then stored on disk. This process in its [...]

Thanks for your kind reply! Also, I would like to know why you use the calibration of kitti dataset instead of dense dataset in precompute.py. Because you have said that the code only implemented the snow simulation of dense dataset

MartinHahner, 2022-08-12:
We use the calibration file from the DENSE dataset.
It is compatible witht the KITTI calibration code.
So we just re-use the KITTI code, but not their calibration.

Rolandxx7, 2022-08-13:
> We use the calibration file from the DENSE dataset. It is compatible witht the KITTI calibration code. So we just re-use the KITTI code, but not their calibration.

Thank you, I see. I met a bug when I was running precompute.py. The reason for the problem is that the variable called ground_distances is empty and the error happens on this file called 2018-02-09_19-10-30_00000.bin. I do not know how to solve this problem because I haven't changed your other code except the corresponding path. Please give me some advice.

`gunn:  63%|██████▎   | 2194/3469 [00:47<00:27, 46.29it/s]
Traceback (most recent call last):
  File "C:/Users/Roland/Desktop/LiDAR_snow_sim-main/tools/snowfall/precompute.py", line 107, in <module>
    beam_divergence=float(np.degrees(3e-3)))
  File "C:\Users\Roland\Desktop\LiDAR_snow_sim-main\tools\snowfall\simulation.py", line 466, in augment
    p = np.polyfit(ground_distances, adaptive_noise_threshold, 2)
  File "<__array_function__ internals>", line 6, in polyfit
  File "C:\Users\Roland\miniconda3\envs\point_rcnn\lib\site-packages\numpy\lib\polynomial.py", line 630, in polyfit
    raise TypeError("expected non-empty vector for x")
TypeError: expected non-empty vector for x`

closed 2022-08-15 by Rolandxx7