SysCV/LiDAR_snow_sim #13: TypeError: expected non-empty vector for x

Rolandxx7, 2022-08-15:
Dear Martin, I met a bug when I was running precompute.py. The reason for the problem is that the variable called ground_distances is empty and the error happens on this file called 2018-02-09_19-10-30_00000.bin. I do not know how to solve this problem because I haven't changed your other code except the corresponding path. Please give me some advice.

`gunn: 63%|██████▎ | 2194/3469 [00:47<00:27, 46.29it/s] Traceback (most recent call last): File "C:/Users/Roland/Desktop/LiDAR_snow_sim-main/tools/snowfall/precompute.py", line 107, in <module> beam_divergence=float(np.degrees(3e-3))) File "C:\Users\Roland\Desktop\LiDAR_snow_sim-main\tools\snowfall\simulation.py", line 466, in augment p = np.polyfit(ground_distances, adaptive_noise_threshold, 2) File "<__array_function__ internals>", line 6, in polyfit File "C:\Users\Roland\miniconda3\envs\point_rcnn\lib\site-packages\numpy\lib\polynomial.py", line 630, in polyfit raise TypeError("expected non-empty vector for x") TypeError: expected non-empty vector for x`

MartinHahner, 2022-08-15:
Probably there is something wrong with the .bin file. 
You could investigate the .bin file with the help of [pointcloud_viewer.py](https://github.com/SysCV/LiDAR_snow_sim/blob/main/pointcloud_viewer.py) or just skip it and any other file that leads to a crash. 
If there are not too many files that leads to a crash, you should still be fine.

closed 2022-08-16 by Rolandxx7

Raiden-cn, 2023-07-26:
Hi, @Rolandxx7 !
 I am stuck in the same bug. 
How did you resolve it?