SysCV/LiDAR_snow_sim #17: Question about Eq. 6

ShengyuH, 2022-09-19:
hi Martin,  

I have questions regarding Equation 6 in the main paper. I can understand that R^2 comes from the solid angle, transmittance takes into consideration the medium beforehand, I don't understand why it's transmittance squared. Can you elaborate a bit here? Also in the text you mentioned "H C depends on the beam divergence, the overlap of transmitter and receiver described by ξ (R) as well as the transmittance T(R) of the medium through", however, I don't see where does the beam divergence play the role here. I understand beam divergence for two roles: 1. intersect with multiple particles within the cone.    2. adjust the energy that follows Gaussian distribution within the cone. So how does it plays the role in the impulse responses of the optical channel? 

Best,
Shengyu

MartinHahner, 2022-09-20:
Hi Shengyu,

T(R) represents the total **one-way** transmission loss, that's why it is squared.
See equation 8 from Rasshofer et al. [33].
I don't know what you mean by "solid angle", but R stands for _Range_ (a.k.a. distance) and has nothing to do with an angle.

Regarding the impact of beam divergence: 
The bigger the opening angle of the beam is, 
the higher the chance it hits a snowflake.
I think that is all we wanted to convey with that statement, so strictly speaking, 
it should have read "H(R) depends on the beam divergence".
So I understand that this statement before Equation 6 could be a bit misleading.

I hope this helps.
Greets, Martin

ShengyuH, 2022-09-20:
hi Martin,

- Yes R stands for the range, but we have 1 / R^2 in the equation because we have to compute the solid angle viewed from the receiver to the surface and the solid angle is computed as: area / R^2, right? 
- Thank for elaboration, now "H(R) depends on the beam divergence" makes sense to me. 

Best,
Shengyu

ShengyuH, 2022-09-20:
I have an additional question: 

- Does the conversion from Eq. 3 to Eq. 11 imply that the transmittance T(R) = 1 for all the evaluated particles / surface? This is because you can pre-compute the occlusion information for the snow flakes and only evaluate the snow flakes that are not occluded?

MartinHahner, 2022-09-20:
> * we have 1 / R^2 in the equation because we have to compute the solid angle viewed from the receiver to the surface and the solid angle is computed as: area / R^2, right?

I have not thought about it from that perspective yet, but your observation is right.

> * Does the conversion from Eq. 3 to Eq. 11 imply that the transmittance T(R) = 1 for all the evaluated particles / surface?

No, T(R) represents the total one-way transmission loss in "open space". 
_The transmittance T(R) is equal to 1 in the part of the medium that is not occupied by snow particles, 
assuming absence of other scatterers._ 

> * This is because you can pre-compute the occlusion information for the snow flakes and only evaluate the snow flakes that are not occluded?

We assume that besides (our sampled) snowflakes, 
there is nothing else in the atmosphere that could interfere with the laser beam.

ShengyuH, 2022-09-20:
```
The transmittance T(R) is equal to 1 in the part of the medium that is not occupied by snow particles,
assuming absence of other scatterers.
```
Yes this is what I was saying: All the **evaluated** snowflakes are not occluded by other particles along the ray. For each evaluated snowflake at range R, the transmittance T(R -\epsilon)=1.

MartinHahner, 2022-09-20:
Yes.

closed 2022-09-20 by ShengyuH