SysCV/LiDAR_snow_sim #18: Dataset

ruveydayilmaz0, 2022-11-28:
Hello, 

Are snow and wet augmentations only available for the DENSE dataset?

dinghuixiang-github, 2022-11-29:
I have the same question. I want to use the model for KITTI dataset in which each point  has 4 channels(x,y,z,intensity) rather than 5 and I don't know to modify the code to achieve the simulation.

> Hello,
> 
> Are snow and wet augmentations only available for the DENSE dataset?

MartinHahner, 2022-11-29:
> Are snow and wet augmentations only available for the DENSE dataset?

Currently, the code needs for every point `[x, y, z, intensity, LiDAR channel/layer]`.
For ideas, how to adapt the code to different datasets, please see [here](https://github.com/SysCV/LiDAR_snow_sim/issues/7#issuecomment-1177903269).

@MarioBijelic: Do you have access to code that can geometrically "reconstruct" the channel information from x, y, z (for KITTI)?

closed 2023-01-13 by github-actions[bot]

gabigarmus, 2024-06-04:
I have the same problem. Would be good if someone had found the solution for that.