SysCV/LiDAR_snow_sim #20: Excuse me, where is the path to save the point cloud file after adding snowflake simulation noise? Thank you！

RhythmOfTheRain-Byte, 2023-02-21:
Excuse me, where is the path to save the point cloud file after adding snowflake simulation noise? Thank you！

MartinHahner, 2023-02-23:
Hi (I guess Pengfei from the email),
you could write your own export function to save the point cloud (`pc`) in your desired format and call that function at the end of `show_pointcloud`, e.g. here:
https://github.com/SysCV/LiDAR_snow_sim/blob/main/pointcloud_viewer.py#L3017
Greets, Martin

closed 2023-02-23 by MartinHahner

RhythmOfTheRain-Byte, 2023-04-25:
Thank you very much！

RhythmOfTheRain-Byte, 2023-07-17:
> Hi (I guess Pengfei from the email), you could write your own export function to save the point cloud (`pc`) in your desired format and call that function at the end of `show_pointcloud`, e.g. here: https://github.com/SysCV/LiDAR_snow_sim/blob/main/pointcloud_viewer.py#L3017 Greets, Martin

Thanks for your time!
In the process of saving the data, I just saved the XYZI 4D information of the point cloud data, which were saved as a bin file.
The size of the saved KITTI single frame data is 16k, which is much smaller than the 1.*M of the original KITTI dataset, I would like to know , will this affect the subsequent training?

esther-ET, 2024-03-02:
> Excuse me, where is the path to save the point cloud file after adding snowflake simulation noise? Thank you！

Hi， may I know how do you get the simulation of snow+wet and which .py should I run to achieve? Thank you very much!