SysCV/LiDAR_snow_sim #22: scatter points

Rolandxx7, 2023-05-08:
Hi @MartinHahner 

The paper seems to only consider the scatter points (raindrops) generated between the object point and the origin(similar to LISA), but raindrops(scatter points) generated in open spaces are not considered. 
So, I would like to know whether raindrop points can be still generated if they are in a very open area?

MartinHahner, 2023-05-08:
Hi @Rolandxx7, 
yes, you are right, we did not consider this case.
Although, I do not know how often there are "missing" measurements under clear conditions.
I assume this case is so rare that for our experiments it would not make a significant difference.

But if you need to cover this case, too, you can extend our code base by looking at the beam angles of every single point in the clear weather point cloud to figure out if there are such missing measurements.

Rolandxx7, 2023-05-09:
Hi @MartinHahner,

Thanks for your quick reply.

I am sorry I do not fully understand this sentence:  "By looking at the beam angles of every single point in the clear weather point cloud to figure out if there are such missing measurements."
1. Why clear weather? Shouldn't we be concerned about bad weather？
2. Does every single point in your sentence represent an object point？

closed 2023-06-23 by github-actions[bot]