SysCV/LiDAR_snow_sim #23: The label of snow point cloud

Grandzxw, 2023-06-14:
I have a question, in the function “process_single_channel”，it is understandable that pc[j, 4] = 1 or pc[j, 4] = 2 can be defined as a snowflake point, while pc[j, 4] = 0 is a normal point? right?

And then I find in wads datasets and cadcd datasets, The intensity of most snowflake  points is 0(zero), So from your simulation, most snowflake point is not 0(zero). Can you explain my question?

Grandzxw, 2023-06-14:
And the function is in the simulation.py. The link is https://github.com/SysCV/LiDAR_snow_sim/blob/760c42e46e10c6aa6b57048c5c1892289f0c2d08/tools/snowfall/simulation.py#L50

MartinHahner, 2023-06-16:
> I have a question, in the function “process_single_channel”，it is understandable that pc[j, 4] = 1 or pc[j, 4] = 2 can be defined as a snowflake point, while pc[j, 4] = 0 is a normal point? right?

```python
pc[j, 4] = 0 # same as original point
pc[j, 4] = 1 # only alter intensity (meaning the location [x,y,z] stays the same
pc[j, 4] = 2 # replace point of hard target with snowflake, meaning not also the intensity, but also the location changes
```

We use these labels for this visualization in the paper:
[image]

> And then I find in wads datasets and cadcd datasets, The intensity of most snowflake points is 0(zero), So from your simulation, most snowflake point is not 0(zero). Can you explain my question?

0 is also a valid intensity, it is the lowest measurable intensity possible by the sensor, it is basically zero-indexed.
Our code was designed to work with the SeeingThroughFog dataset, it is possible, that some parameters (e.g. [here](https://github.com/SysCV/LiDAR_snow_sim/blob/760c42e46e10c6aa6b57048c5c1892289f0c2d08/tools/snowfall/simulation.py#L72-L76) and [there](https://github.com/SysCV/LiDAR_snow_sim/blob/760c42e46e10c6aa6b57048c5c1892289f0c2d08/tools/snowfall/simulation.py#L106-L109)) need to be adjusted to work well with other datasets while the overall algorithm should still work.

closed 2023-06-26 by Grandzxw