SysCV/LiDAR_snow_sim #24: Error while cloning the repo !

ammaryasirnaich, 2023-06-22:
Hi really appreciate the work that you have shared. I was trying to give it a try but getting an error while cloning the repo
```
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.
```
even doing the `sudo clone ` i am still getting errors related to access rights 

```
Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:MartinHahner/LISA' into submodule path '/home/ammar/lidar_simulator/LiDAR_snow_sim/lib/LISA' failed
Failed to clone 'lib/LISA'. Retry scheduled
Cloning into '/home/ammar/lidar_simulator/LiDAR_snow_sim/lib/LiDAR_fog_sim'...
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:MartinHahner/LiDAR_fog_sim' into submodule path '/home/ammar/lidar_simulator/LiDAR_snow_sim/lib/LiDAR_fog_sim' failed
Failed to clone 'lib/LiDAR_fog_sim'. Retry scheduled
Cloning into '/home/ammar/lidar_simulator/LiDAR_snow_sim/lib/OpenPCDet'...
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:MartinHahner/OpenPCDet' into submodule path '/home/ammar/lidar_simulator/LiDAR_snow_sim/lib/OpenPCDet' failed
```

Would appreciate if i can get help

AlanLiangC, 2023-07-06:
Hi, I have the same problem, have you solved?

MartinHahner, 2023-07-06:
https://stackoverflow.com/a/12940681/5324725